JdeRobot/drones #16: Adapt drone_teleop to show position and velocities info

pariaspe, 2020-04-24:


diegomrt, 2020-05-04:
Hi @pariaspe. Just checked your new drone_teleop GUI proposal. Looks good. Just a few suggestions:

- I'll use at least two digits after the decimal point in both pose and velocity.
- I'll add, after the three components, the reference frame in which they're given (maybe the descriptive mav frame would be a good option).
- I'll also include angular velocity
- Maybe showing them clustered and writing complete labels -Pos Info (M) -> Position (m) and same for velocity- would help reading them

Thanks!

pariaspe, 2020-05-05:
Hi @diegomrt , 

Thanks for you feedback. What do you think about this new gui?

[image]

I am not sure where the reference frame is. I am reading it from the header of PoseStamped and TwistStamped msgs (both from local position topic). Is it ok?

diegomrt, 2020-05-05:
Hi @pariaspe 

The new GUI now looks great to me. Maybe I'll just point out somehow that the fourth velocity component is angular z speed, in rad/s. Now it's not clear. Use a new line or insert it between the actual data and the JdeRobot logo. 

Regarding frames, its ok to use the ROS ones. Probably the mav frame has no interest here.

pariaspe, 2020-05-05:
> Hi @pariaspe
>
> The new GUI now looks great to me. Maybe I'll just point out somehow that the fourth velocity component is angular z speed, in rad/s. Now it's not clear. Use a new line or insert it [...]

Nice! What about this one then?

[image]

diegomrt, 2020-05-05:
Perfect @pariaspe, I think we've got it!

commit 4183a62d (2020-05-05): Improvements on new GUI. Closes #16.
closed 2020-05-05 by pariaspe