JdeRobot/drones #21: [drone_teleop] Disable takeoff till autopilot is ready

diegomrt, 2020-04-30:
After lauching, if PX4 isn't fully loaded takeoff button doesn't work, and it remains faulty

pariaspe, 2020-05-04:
Hi @diegomrt,

I've just tried to takeoff immediately after launching and I didn't have any problem. The drone waits until PX4 GPS fusion is completed and after that starts a standard takeoff.

Can you add more information about your bug?

diegomrt, 2020-05-04:
Hi @pariaspe,

I've just tried it again and the faulty behaviour is still there. I'm launching the "gymkhana exercise" and pressing the takeoff button before GPS fusion and before OFFBOARD mode is requested. PX4 Firmware is updated to its last version.

Drone arming and takeoff are detected but the drone remains just armed at the takeoff pad, without actually taking off. And it never does till the exercise is reloaded completely. PX4 keeps armind and disarming cyclically.

The screenshot shows takeoff is detected but the drone is at its pad:

[image]

pariaspe, 2020-05-05:
Hi again @diegomrt ,

I'm not facing this error, neither in the gymkhana exercise. The screenshot shows the execution sequence at the command line. Take off is requested before GPS fusion, and after completing it, the drone takes off immediately.

[image]

Have you checked it on different exercises? And have you checked if there is a bug on your drone_wrapper?

diegomrt, 2020-05-05:
Hi @pariaspe 

I've updated to the last drone_wrapper and the takeoff error is still there, regardless the Academy exercise I launch if I press takeoff too early the drone doesn't move. 

Seems more related to PX4. Could you please update to its last version and try again?

pariaspe, 2020-05-05:
> Hi @pariaspe
> 
> I've updated to the last drone_wrapper and the takeoff error is still there, regardless the Academy exercise I launch if I press takeoff too early the drone doesn't move.
> 
> Seems more related to PX4. Could you please update to its last version and try again?

I've just updated to PX4 last version and no error appears. Maybe we should ask a third person if they also meet this error.

Klego, 2020-05-22:
I have had this error. The problem is if you push the TakeOff button before this message, your dron never takesoff:
[image]
In order to make this work, you have to wait for this message to appears before pushing the button.

closed 2020-06-08 by pariaspe

diegomrt, 2020-06-08:
@pariaspe, how was this finally solved?

pariaspe, 2020-06-08:
> @pariaspe, how was this finally solved?

Just checking if the landed_state is defined when the button is clicked. I used part of the code needed to fix #41.

As this issue didn't show up to me, it will be great if you could check that it works well.

PR #49 (2020-06-08): Takeoff button synchro
Synchronizing between vel and pose teleops. Closes #41.
Also fixes #21. Takeoff is only available when the drone is ready.
commit fb40f6e3 (2020-06-08): Takeoff button disabled until landed state is defined (drone is ready). Closes #21