JdeRobot/drones #38: [drone_teleop] Cameras and pos/vel freeze in new GUI

diegomrt, 2020-05-29:
Each 20/30 sec both the cameras and the position/velocities freeze in the new drone_teleop GUI, as it's seen in the screenshot below. Tested using the new drone_teleop_all.perspective. 

Clicking in some precise points/controls of the GUI unfreeze the perspective temporarily.

[image]

diegomrt, 2020-06-05:
I think we should try if [throttling](http://wiki.ros.org/topic_tools/throttle) the mavros pose and vel topics to rates < 5 Hz (now they publish at 30 Hz) avoids this issue. 

I've checked the cameras perspective alone and it doesn't freeze.

pariaspe, 2020-06-08:
The problem seems to be that the GUI can not update itself as fast as the Subscribers receive msgs with the pose and twist. Then, as you suggest, msgs have to be throttled to reduce the GUI update rate.

This bug could be fixed by creating a Thread which periodically updates the position and velocity info:
```python
threading.Timer(0.5, self.update_status_info).start()
```

I've tried more ROS friendly solutions using [rospy.Timer](http://docs.ros.org/jade/api/rospy/html/rospy.timer.Timer-class.html), [throttle](http://wiki.ros.org/topic_tools/throttle) or [drop](https://wiki.ros.org/topic_tools/drop), but I didn't get any working solution.

diegomrt, 2020-06-08:
Hi @pariaspe, I seems good to me if it works and it doens't overload the CPU. 

In any case, ROS-based solutions could rely on suscribing the GUI to new throttled pose and vel topics, that can be published at least in two different ways:

- Launching a ROS throttling node directly from the exercise .launch (easiest way, a line of code)
- Building our own "republish 1, discard the next 14 messages" solution within the drone_wrapper, which is capable of handling easily the actual 30 Hz callbacks

closed 2020-06-08 by pariaspe

pariaspe, 2020-06-08:
I've just created a new issue to implement a new ROS-based solution that fixes this issue too.

PR #48 (2020-06-08): RQT freezing fixed
Closes #38 and closes #39
commit eb52fcff (2020-06-08): RQT freezing fixed. Closes #38 and closes #39