JdeRobot/drones #39: [drone_teleop] Position in GUI partly synced with mavros topic

diegomrt, 2020-05-29:
Some components of the Position information box of the new GUI freeze and never recover back. Interestingly, some others keep synced with `/mavros/local_position/pose` topic

The screenshot below was taken when x,y were frozen but z was still syncinc properly: 

[image]

pariaspe, 2020-06-08:
Hi @diegomrt ,

This issue seems similar to #38. Would not be the same problem?

diegomrt, 2020-06-08:
Hi @pariaspe 

Yes, I also think this particular behaviour about partial sync in the new box probably comes from the same issue described in #38. So we'll close it once #38 is solved.

PR #48 (2020-06-08): RQT freezing fixed
Closes #38 and closes #39
commit eb52fcff (2020-06-08): RQT freezing fixed. Closes #38 and closes #39
closed 2020-06-08 by pariaspe