JdeRobot/drones #80: [drones] support for IRIS drone in noetic

jmplaza, 2021-04-19:
so it may be used both in RoboticsAcademy and also in VisualCircuit. We will create a noetic-branch

RUFFY-369, 2021-04-21:
Hi @jmplaza ,
I would like to contribute to this issue. Regarding the workflow, should the installation instructions for Ubuntu 20.04 (Focal) release, ROS Noetic be updated first?

Regards,
Prakarsh

pariaspe, 2021-04-21:
Hi @RUFFY-369, 

It would be nice if you create an _**installation20.md**_ following the installation steps of the [previous version](https://github.com/JdeRobot/drones/blob/master/installation18.md).

RUFFY-369, 2021-04-22:
Hi @pariaspe ,
Yeah sure, will work on it.

RUFFY-369, 2021-04-27:
Hi @pariaspe ,
I have created a PR (#84 ) regarding creating an **_installation20.md_**. Please review it.

Regards,
Prakarsh

pariaspe, 2021-04-28:
jderobot_drones has been migrated to noetic ([branch](https://github.com/JdeRobot/drones/tree/noetic-devel)). Installation instructions has been also updated in #84 and #86.
Thanks for your contribution @RUFFY-369, @iamarkaj

closed 2021-04-28 by pariaspe