JdeRobot/drones #87: Run RoboticsAcademy position_control exercise in noetic and gazebo-11

iamarkaj, 2021-04-29:
**I was able to run position_control on Ros-noetic and Gazebo-11 but I need some help to solve few errors.**

A running video is posted below

https://user-images.githubusercontent.com/40225765/116550348-48c14d80-a914-11eb-9fd2-03da5aa82608.mp4

**Errors**
- **Error when running the below command to update ros dependencies**
```
rosdep install --from-paths . --ignore-src --rosdistro noetic -y
```
[image]

*For now, I skipped that*

- **Path errors**
[image]

*For now, added the below in bashrc*
```
export PYTHONPATH=$PYTHONPATH:~/catkin_ws/src/rqt_drone_teleop/src/rqt_vel_teleop:~/catkin_ws/src/drone_wrapper/src/drone_wrapper
```
Also, attaching the image of my bashrc
[image]

- **rqt_gui error which can already be seen in the video**
[image]

*For now, running the below in a separate window*
```
rosrun rqt_gui rqt_gui
```

iamarkaj, 2021-04-29:
**Update: fixed the rqt_gui problem**
Adding "--" before perspective file args inside the launch file fixed it.

Inside position_control.launch
```
<node name="rqt_gui" pkg="rqt_gui" type="rqt_gui" respawn="false" output="screen" args="perspective-file $(arg perspective)"/>
```
to
```
<node name="rqt_gui" pkg="rqt_gui" type="rqt_gui" respawn="false" output="screen" args="--perspective-file $(arg perspective)"/>
```

RUFFY-369, 2021-05-01:
Hi @iamarkaj , @pariaspe ,
This is in reference to the error caused by executing the following command:
`rosdep install --from-paths . --ignore-src --rosdistro noetic -y`
which @iamarkaj mentioned.

I looked into it and found that there are some distribution specific dependencies of Ubuntu releases and seems that python-rosdep doesn't exist in Ubuntu 20.04, i.e., Focal. Here is the reference to the [TypeScript](https://github.com/ros-tooling/setup-ros/blob/master/src/package_manager/apt.ts#L37) file which mentions it. And they metioned a [pull](https://github.com/ros-tooling/setup-ros/pull/147/files) which they already created regarding it.

The above installation (as shown in image) failed as ROS Noetic utilises python3 unlike Melodic which uses python2.
Did you (@iamarkaj) already install the python/python3-rospkg and is there any package on your system with python-rospkg as dependency instead of python3-rospkg?

iamarkaj, 2021-05-01:
[image]

Do let me know if this helps @RUFFY-369

RUFFY-369, 2021-05-02:
Hi @iamarkaj ,
Can you try to execute the `rosdep install --from-paths . --ignore-src --rosdistro noetic -y` from a different directory instead of catkin_ws? And tell if 'python-rospkg' installation error still occurs.
And also is python3 your default python?

iamarkaj, 2021-05-02:
Hi @RUFFY-369 
AFAIK, we need to make sure we are inside catkin workspace for rosdep to look at all the packages in the src directory and find all of the dependencies they have. Then it will recursively install the dependencies. Correct me if I am wrong.

And for the python version, I am attaching a ss.
[image]

RUFFY-369, 2021-05-02:
Hi @iamarkaj ,
The reason  for asking to rosdep install from another directory was to confirm that is there any package in your workspace with python-rospkg as  a dependency which is causing the error. 
And regarding  being in catkin_ws for rosdep to find package dependencies, one can try it from a different catkin_ws to confirm the same.

pariaspe, 2021-05-02:
Hi guys,
I've been working on a new docker image based on noetic lately, so I hope my little experience with noetic can help. I got some errors configuring my catkin_ws, but I solved them installing **python3-catkin-tools** and **python3-rosdep**. Notice that python2 (and pkgs related to it) are not needed anymore with Ubuntu 20.04 and ROS Noetic.
Rosdep is used to check if all ROS pkgs have their required dependencies already installed in the host machine. It is used with source packages, so it has to be run inside the catkin_ws folder.

PS: I hope to review (and merge) all your PRs soon. Sorry for the delay.

iamarkaj, 2021-05-04:
Hi @pariaspe 
Can I work on the RoboticsAcademy Docker image or is it done?

pariaspe, 2021-05-04:
> Hi @pariaspe
> Can I work on the RoboticsAcademy Docker image or is it done?

It is almost done, I hope finish testing and pushing it soon.

pariaspe, 2021-06-16:
Soled in JdeRobot/RoboticsAcademy#1027

closed 2021-06-16 by pariaspe