JdeRobot/drones #92: Update README with new noetic branch

pariaspe, 2021-05-04:
Current changes in master branch (#90) have caused a jenkins build [fail](https://build.ros.org/job/Kdev__jderobot_drones__ubuntu_xenial_amd64/50/console). It seems that gazebo pkg it is not found in drone_assets. 
Since Kinetic branch is on EOL status, and new changes won't be included in kinetic release, I propose to update README removing kinetic tags and adding noetic new ones.

Drone_assets tags should be also added.

jmplaza, 2021-05-04:
perfect, good forward movement

pariaspe, 2021-05-04:
Bionic jenkins build has also [failed](https://build.ros.org/job/Mdev__jderobot_drones__ubuntu_bionic_amd64/38/console). Drone_assets' CMakeLists.txt have to be reviewed as it is raising the error.

iamarkaj, 2021-05-04:
PR #93

closed 2021-05-09 by pariaspe