JdeRobot/drones #97: Issue with installation on Noetic.

AbsorbedInThought, 2021-05-05:
I'm trying to get the follow_road exercise working on ROS Noetic on Ubuntu 20.04 but I am getting this error:

> taha@taha-PC:~/Academy/exercises/static/exercises/follow_road$ roslaunch follow_road.launch
> ... logging to /home/taha/.ros/log/257ff334-ad9d-11eb-ad94-251a44ff69c3/roslaunch-taha-PC-65169.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> Resource not found: The following package was not found in <arg name="perspective" default="$(find rqt_drone_teleop)/perspectives/drone_teleop_vel_cam.perspective"/>: rqt_drone_teleop
> ROS path [0]=/opt/ros/noetic/share/ros
> ROS path [1]=/opt/ros/noetic/share
> The traceback for the exception was written to the log file

This file "rqt_drone_teleop)/perspectives/drone_teleop_vel_cam.perspective" file is present and it seems to be a problem with environment variables. However I have double checked them and can't figure out what's wrong.
I have also attached a file that contains all of the output generated by the terminal during installation. 
[installation.txt](https://github.com/JdeRobot/drones/files/6427075/installation.txt)

pariaspe, 2021-05-05:
Hi @AbsorbedInThought,
It seems that your $ROS_PACKAGE_PATH is not well configured. Make sure that you have included _source ~/catkin_ws/devel/setup.bash_ to your _.bashrc_

AbsorbedInThought, 2021-05-05:
I did configure that path multiple times but I was still getting the same error. After running roscd && catkin build on the directory, it displayed a message saying: "[build] Note: Workspace packages have changed, please re-source setup files to use them." I did source it after it displayed this but it still didn't work.

iamarkaj, 2021-05-05:
@AbsorbedInThought I am not sure if this can help you but here is my bashrc
```
# ROS-noetic
source /opt/ros/noetic/setup.bash
source /usr/share/gazebo/setup.sh
source ~/catkin_ws/devel/setup.bash

export ROS_WORKSPACE=~/catkin_ws
export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:/usr/share/gazebo-11
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:~/repos/PX4-Autopilot/build/px4_sitl_default/build_gazebo
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/repos/PX4-Autopilot/Tools/sitl_gazebo/models:~/repos/drones/drone_assets/models
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/repos/PX4-Autopilot/build/px4_sitl_default/build_gazebo

export PYTHONPATH=$PYTHONPATH:~/catkin_ws/src/rqt_drone_teleop/src/rqt_vel_teleop:~/catkin_ws/src/drone_wrapper/src/drone_wrapper:~/catkin_ws/src/rqt_ground_robot_teleop/src/rqt_ground_robot_teleop

export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/repos/PX4-Autopilot
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/repos/PX4-Autopilot/Tools/sitl_gazebo
```

pariaspe, 2021-05-06:
I'm posting mine as well. It uses ROS Melodic, but it might help you too.

```bash
## Install ROS
source /opt/ros/melodic/setup.bash
source ~/catkin_ws/devel/setup.bash
export ROS_WORKSPACE=~/catkin_ws
export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:/usr/share/gazebo-9
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:~/jderobot/drones/drone_assets/models

## Install PX4
# setup Gazebo env and update package path
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:~/repos/Firmware/build/px4_sitl_default/build_gazebo
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/repos/Firmware/Tools/sitl_gazebo/models
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/repos/Firmware/build/px4_sitl_default/build_gazebo

export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/repos/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:~/repos/Firmware/Tools/sitl_gazebo
```

pariaspe, 2021-05-12:
Hi @AbsorbedInThought, have you managed to fix the installation problem?

AbsorbedInThought, 2021-05-13:
I did verify my bashrc with yours and it seems to be properly configured. I think it might be an issue with something else. Some things in the installation went different for me, like rosdep init failed so I had to install rosdep2. I also read rosdep2 docs that said it is currently unstable right now. I haven't really been able to play around with the installation but I'll update you in a 2-4 days because I'm on Eid holidays right now.

AbsorbedInThought, 2021-05-18:
@pariaspe I haven't been able to fix it completely but there is some progress.

Firstly, I had to explicitly add this to bashrc (I don't think the installation contains this step):
`export PYTHONPATH=$PYTHONPATH:~/catkin_ws/src/rqt_drone_teleop/src/rqt_vel_teleop:~/catkin_ws/src/drone_wrapper/src/drone_wrapper:~/catkin_ws/src/rqt_ground_robot_teleop/src/rqt_ground_robot_teleop`

It didn't work after that so I created a new catkin workspace. Also, I think it is better to update your ROS installation after this installation process "sudo apt upgrade ros-noetic-desktop-full". It basically downloaded some rqt packages which I think were missing in my installation.

However, I am getting this issue right now:
[image]

pariaspe, 2021-05-18:
Hi @AbsorbedInThought ,

Correct me If I am wrong, but the installation instructions actually recommend to install `ros-noetic-desktop-full` ([installation20](https://github.com/JdeRobot/drones/blob/master/installation20.md)).

You can fix that error installing jderobot QFlightInstruments (qfi) module. You can find it at [JdeRobot/ThirdParty](https://github.com/JdeRobot/ThirdParty/tree/master/qflightinstruments) repo.

AbsorbedInThought, 2021-05-18:
I basically had noetic installed already so I skipped that step. I think if someone skips that step then he should probably try upgrading noetic. Maybe we can add that to the troubleshooting section.

Thanks for providing that module, I just couldn't find it anywhere.

AbsorbedInThought, 2021-05-30:
[image]

I did get it working but for some reason, the drone works fine but isn't displayed correctly.

pariaspe, 2021-05-30:
Hi @AbsorbedInThought,

That's really weird. What may happen is that gazebo can find where the model is. Have you included drone_assets location to GAZEBO_MODEL_PATH?

AbsorbedInThought, 2021-05-30:
Yes, I have. For some reason, gazebo takes a long time to load this particular simulation. Usually around 2-3 mins. On PX4, it works fine.

pariaspe, 2021-05-30:
> Yes, I have. For some reason, gazebo takes a long time to load this particular simulation. Usually around 2-3 mins. On PX4, it works fine.

That may be because gazebo looks for the models on its server, rather than looking for them locally. Downloading those models is what takes so long.

How do you install drones packages? Binaries are available since friday [[1]](https://discourse.ros.org/t/new-packages-for-noetic-2021-05-28/20652) [[2]](http://repositories.ros.org/status_page/ros_noetic_default.html?q=jderobot_drones).

AbsorbedInThought, 2021-05-30:
I used the source installation. The binary installation had some errors connecting to the server. But I think it has something to do with my download region. I'll change my download region and try the binary installation.

pariaspe, 2021-05-30:
It have to be a problem with the environment variables. Can you please copy me your `.bashrc`?

I haven't tested the noetic binaries yet, but it shouldn't have problems.

iamarkaj, 2021-05-31:
> [image]
> 
> I did get it working but for some reason, the drone works fine but isn't displayed correctly.

This PR #106 will fix this

pariaspe, 2021-05-31:
Hi @AbsorbedInThought , 

@iamarkaj has added the drone meshes to the package, could you tried again and check if that patch fix the issue?

pariaspe, 2021-06-17:
Stale, closing in a couple of days if no new updates occur

AbsorbedInThought, 2021-06-17:
The PR #106 fixed the issue with the drone and now it works fine.

closed 2021-06-17 by AbsorbedInThought