JdeRobot/drones #141: drone_wrapper build failing due to drone_tello_class

nkhedekar, 2021-09-23:
Build failure log output: https://build.ros.org/job/Nbin_ufhf_uFhf__drone_wrapper__ubuntu_focal_armhf__binary/18/console

Main issue :
```bash
23:16:26 CMake Error at cmake_install.cmake:60 (file):
23:16:26   file INSTALL cannot find
23:16:26   "/tmp/binarydeb/ros-noetic-drone-wrapper-1.4.1/src/drone_tello_class.py":
23:16:26   No such file or directory.
23:16:26 
```

pariaspe, 2021-09-23:
On `noetic_devel`:
https://github.com/JdeRobot/drones/blob/7bfac62d134ce58f40f92de760f0823ff2d170f7/drone_wrapper/CMakeLists.txt#L19

On `melodic_devel`:
https://github.com/JdeRobot/drones/blob/754b95de3b1fe8fbd80be275a2d4f993b44099d9/drone_wrapper/CMakeLists.txt#L19

That file have to be removed from CMakeLists, it doesn't exist anymore. I'm quite busy these days, sorry for all the Jenkins Build mails @nkhedekar

pariaspe, 2021-11-09:
Fixed in #142 #143 #147

PR #147 (2021-11-04): [melodic] Fix jenkins build on drone_assets package
Adding gazebo_ros dependency to package.xml
Closes #141
closed 2021-11-09 by pariaspe