JdeRobot/drones #179: jderobot-assets clashes with drone-circuit-assets (ubuntu install)

gavanderhoorn, 2022-08-24:
I'm not a user of these packages, but during a scripted installation of *all* released ROS Noetic packages, I ran into the following problem:

```
Unpacking ros-noetic-jderobot-assets (1.1.0-1focal.20210423.224520) ...
dpkg: error processing archive /tmp/apt-dpkg-install-P8q7sZ/2501-ros-noetic-jderobot-assets_1.1.0-1focal.20210423.224520_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/noetic/gazebo/assets-setup.sh', which is also in package ros-noetic-drone-circuit-assets 1.4.2-1focal.20211004.214250
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
```

note the `trying to overwrite '/opt/ros/noetic/gazebo/assets-setup.sh', which is also in package ros-noetic-drone-circuit-assets 1.4.2-1focal.20211004.214250` part.

It appears two released packages are trying to install the same file, in the same location.

If users should not try to install both at the same time, perhaps a [breaks](https://www.debian.org/doc/debian-policy/ch-relationships.html#packages-which-break-other-packages-breaks) relation should be declared, or if desirable, a `conflicts` relation.

It would be great if this could be solved, as this conflict prevents `apt` from successfully installing packages.

pariaspe, 2022-08-24:
Hi @gavanderhoorn , thanks for noticing. `jderobot-assets` pkg is deprecated in favor of `drone-circuit-assets` and other packages. We'll try to fix the conflict asap.

gavanderhoorn, 2022-09-08:
Friendly ping?

gavanderhoorn, 2022-10-25:
Another friendly ping.

closed 2024-08-16 by gavanderhoorn